RobotWebTools/rclnodejs #924: Electron Renderer Process

KhalilSelyan, 2023-06-23:
**Description**
It seems as though you can only access rclnodejs on the electron main process (node backend) and not on the frontend the renderer process. Is there any way to get it running directly from the frontend ?

- Library Version: 0.22.2
- ROS Version: humble
- Platform / OS: ubuntu 22.04

**Steps To Reproduce**

create a electron project and add rclnodejs and try to access it in the front end 
because of electron updates since i believe electron 11 "non context-aware native modules are not allowed"
i need to be able to run the native connection directly from the frontend because of buffering issues when trying to send from main process to renderer process because pointclouds and image messages are too big 

[image]

**Expected Behavior**

**Actual Behavior**

minggangw, 2023-06-25:
Based on my limited knowledge about Electron, this is how Electron supports nodejs modules. Please see https://www.electronjs.org/docs/latest/tutorial/process-model#the-main-process
[image]

Tombliboo96, 2024-05-10:
@KhalilSelyan Hello, may I ask how you dealt with this problem? I encountered the same problem as you

KhalilSelyan, 2024-05-10:
I moved away from rclnodejs and used ros-foxglove-bridge with ws-protocol both from foxglove

minggangw, 2024-05-11:
Hi @KhalilSelyan, how about the performance using ros-bridge? As I understand, the current `ros-bridge` depends on websocket and needs to copy the data either, I'm trying to figure out if it's possible to avoid the memory allocation using web technology, or you leverage other tech, like video streaming, to solve the problem? thanks!

KhalilSelyan, 2024-05-11:
i remember the issues i had with the ros-bridge was that it was slow and couldn't handle big pointcloud messages + other stuff -- i'm working with the autoware software -- although after finding the foxglove rosbridge it was much better at handling all those with barely any lag , still not as perfect as a native connection i would assume but definitely good enough if configured correctly in my opinion, but for your other question i'm not entirely sure how to answer them sorry

minggangw, 2024-09-26:
@KhalilSelyan please check with rclnodejs Electron demo just added https://github.com/RobotWebTools/rclnodejs/tree/develop/electron_demo