RobotWebTools/rclnodejs #937: autofill JointState msg header when object is created

kuriousDesign, 2023-10-04:
Pretty new to submitting features requests so please be patient!

I am developing an app with react frontend that with communicates with a backend over socketio. The backend connects to ROS2 using rclnodejs. 

My feature request involves auto-populating the header data in the JointState msg object. I cannot get my code to work without autofilling the header data as shown below:

      const pub_ = this.node.createPublisher(RosMsgTypes.JointState, 'topic_js');
      const stringMsgObject = rclnodejs.createMessageObject(RosMsgTypes.JointState);
      stringMsgObject.header.stamp.sec = 0.0;
      stringMsgObject.header.stamp.nanosec = 0.0;
      stringMsgObject.header.frame_id = '0';

minggangw, 2023-10-05:
@kuriousDesign thanks for your feedback, I will take a look after my holidays.

minggangw, 2023-10-25:
@kuriousDesign sorry for the delay, I didi some testing and it seems that removing the validation can meet your requirement
https://github.com/RobotWebTools/rclnodejs/blob/795e36dfb22b5f2911b7275f0f71e597a2ca4eaa/rosidl_gen/templates/message.dot#L368-L370

But I haven't done more testing to check if any side-effect.

PR #947 (2024-02-04): Add willCheckConsistency option to enable/disable message consistency checking
Currently, when a publisher/client sends a topic/request, it will check the consistency of the message to be sent. For example, the following code will lead to an exception:

```js
  const pub_ = this.node.createPublisher('sensor_msgs/msg/JointState', 'topic_js');
  const stringMsgObject = rclnodejs.createMessageObject('sensor_msgs/msg/JointState');
  stringMsgObject.header.frame_id = '0';
```
Because the `stamp` in `header` is not initialized.

This patch adds an additional option for node when initializing, which  [... cut]
commit e4ce8e69 (2024-02-04): Add willCheckConsistency option to enable/disable message consistency checking (#947)

Currently, when a publisher/client sends a topic/request, it will check the consistency of the message to be sent. For example, the following code will lead to an exception:

```js
  const pub_ = this.node.createPublisher('sensor_msgs/msg/JointState', 'topic_js');
  const stringMsgObject = rclnodejs.createMessageObject('sensor_msgs/msg/JointState');
  stringMsgObject.header.frame_id = '0';
```
Because the `stamp` in `header` is not initialized.

This patch adds an additional option for node when initializing [... cut]
commit c4c590a4 (2024-02-04): Add willCheckConsistency option to enable/disable message consistency checking (#947)

Currently, when a publisher/client sends a topic/request, it will check the consistency of the message to be sent. For example, the following code will lead to an exception:

```js
  const pub_ = this.node.createPublisher('sensor_msgs/msg/JointState', 'topic_js');
  const stringMsgObject = rclnodejs.createMessageObject('sensor_msgs/msg/JointState');
  stringMsgObject.header.frame_id = '0';
```
Because the `stamp` in `header` is not initialized.

This patch adds an additional option for node when initializing [... cut]
closed 2024-02-04 by minggangw