RobotWebTools/rclnodejs #943: librcl.so: cannot open shared object file: No such file or directory

Mithilesh982, 2023-11-24:
**Description**
I'm built a web app using React as frontend and node.js as backend  , to connect ros2 and my nodejs im using rclnodejs but while running the code error is showing  like : 
server/node_modules/bindings/bindings.js:121
        throw e;
        ^
Error: librcl.so: cannot open shared object file: No such file or directory
    at Module._extensions..node (node:internal/modules/cjs/loader:1473:18)
    at Module.load (node:internal/modules/cjs/loader:1207:32)
    at Module._load (node:internal/modules/cjs/loader:1023:12)
    at Module.require (node:internal/modules/cjs/loader:1235:19)
    at require (node:internal/modules/helpers:176:18)
    at bindings (/home/eric/CUB-E_frontent_r2/server/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/home/eric/CUB-E_frontent_r2/server/node_modules/rclnodejs/lib/clock.js:17:38)
    at Module._compile (node:internal/modules/cjs/loader:1376:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1435:10)
    at Module.load (node:internal/modules/cjs/loader:1207:32) {
  code: 'ERR_DLOPEN_FAILED'
}
-node : v20.10.0
- Library Version: rclnodejs 0.23.2
- ROS Version: Ros2 humble
- Platform / OS: ubuntu 22.04

**Steps To Reproduce**

const rclnodejs = require('rclnodejs')
rclnodejs.init().then(()=>{
    const node = new rclnodejs.Node('ros2_connectivity_turtlebot3');
    const publisher = rclnodejs.createPublisher('std_msgs/msg/String','cmd_vel ')
    publisher.publish('Hello from react server')

    node.spin()
})

**Expected Behavior**
srever running on 4000

**Actual Behavior**
error

minggangw, 2023-12-01:
I'm using nodejs v20.9.0, but didn't meet this error. Not sure why it doesn't work on your env.

Tombliboo96, 2024-04-19:
Ok, thanks, I'll try again.

minggangw, 2025-06-30:
Close this issue as no more feedback, please try with latest rclnodejs release to see if it can repro.

closed 2025-06-30 by minggangw