RobotWebTools/rclnodejs #948: Node.JS Version

gokhangunduz, 2023-12-28:
Hello, first of all, thank you very much for both developing and maintaining such a project. It's a project that I'm really happy to see and use!

Due to the bug, [Node.JS](https://nodejs.org/en) has now increased to 20.10.0 as the LTS version as of today. I don't know if support for 
the 18.X version should be discontinued, but I think 20.X support should definitely come.

Thanks in advance for everything. Best wishes.

[image]

chfritz, 2024-03-28:
My understanding is that as off [v0.25](https://github.com/RobotWebTools/rclnodejs/releases/tag/0.25.0) rclnodejs *does* support  Node 20. We use it in many [Transitive](https://github.com/transitiverobotics/transitive) capabilities, and have been testing with Node 20 as of late. Is there a specific issue you saw in Node 20, @gokhangunduz ?

minggangw, 2024-03-29:
Hi @chfritz  @gokhangunduz Our goal is to support as many nood.js versions as possible, considering the fragmentation of it, and we will drop the versions that don't support kind of new features of JS we leverage. Supporting the active LTS is our priority definitely. You can see the current CI runs on:

- [Windows platform](https://github.com/RobotWebTools/rclnodejs/blob/develop/.github/workflows/windows-build-and-test.yml) for Node.js 18 & 20
- [Linux platform](https://github.com/RobotWebTools/rclnodejs/blob/develop/.github/workflows/linux-build-and-test.yml) for Node.js 18 & 20

So if you find anything wrong when running `rclnodejs` with latest LTS, please open an issue here, thanks!

minggangw, 2024-04-03:
Close this issue.

closed 2024-04-03 by minggangw