RobotWebTools/rclnodejs #949: Publisher topic name field not accurately giving the topic name.

PranavDhulipala, 2024-01-04:
When you create a publisher using 
`let publisher = node.createPublisher('std_msgs/msg/String', 'test/this/chatter');`
and then do `console.log(publisher.topic);`
it prints only chatter and not test/this/chatter but it seems to publish on  test/this/chatter though

- Library Version: "rclnodejs": "^0.23.3"
- ROS Version: Humble
- Platform / OS: Ubuntu 22.0.4

**Steps To Reproduce**
```
import * as rclnodejs from 'rclnodejs';

// Create a node that publishes a msg to the topic 'foo' every 1 second.
// View the topic from the ros2 commandline as shown below:
//     ros2 topic echo foo std_msgs/msg/String
async function example() {

  await rclnodejs.init();
  let node = rclnodejs.createNode('MyNode');
  
  // Create main working components here, e.g., publisher, subscriber, service, client, action
  // For example, a publisher sending a msg every 1 sec
  let publisher = node.createPublisher('std_msgs/msg/String', 'test/this/chatter');
  console.log(publisher.topic);
  let cnt = 0;
  let msg = rclnodejs.createMessageObject('std_msgs/msg/String');
  node.createTimer(1000, () => {
    msg.data = `msg: ${cnt += 1}`
    publisher.publish(msg);
  });

  node.spin();

}

(async function main(): Promise<void> {
  example();
})().catch((): void => {
  process.exitCode = 1
})
```

it prints only chatter and not test/this/chatter

minggangw, 2024-01-05:
@PranavDhulipala Thanks for submitting the issue, I'm going to take a look later.

closed 2024-01-08 by minggangw

minggangw, 2024-01-08:
@PranavDhulipala Please check with latest https://www.npmjs.com/package/rclnodejs/v/0.24.0, now the behavior aligns with rclpy.

PranavDhulipala, 2024-01-08:
@minggangw thanks for the fix, this works. However the behavior is similar for createService . I can open another bug if you want me to

minggangw, 2024-01-09:
Thanks for your feedback, I'm going to open an issue to track the client/service problem. @PranavDhulipala

minggangw, 2024-01-09:
> @minggangw thanks for the fix, this works. However the behavior is similar for createService . I can open another bug if you want me to

#952 created

minggangw, 2024-01-10:
@PranavDhulipala please try with https://www.npmjs.com/package/rclnodejs/v/0.25.0 and feel free to open issue if you find, you are welcome to contribute!

PranavDhulipala, 2024-01-10:
Confirmed, it works thanks :)

PR #951 (2024-01-08): Leverage rcl functions to get topic of publisher/subscription
This patch implements:

1. Return the result of rcl_publisher_get_topic_name() directly, instead of splitting it by "/" and returns the last part.
2. Use rcl_subscription_get_topic_name() to get the topic of a subscription.

Meanwhile, this patch updates the following tests:

- test/test-node-oo.js
- test/test-node.js
- test/test-remapping.js

Fix: #949, #950
commit 625b016f (2024-01-08): Leverage rcl functions to get topic of publisher/subscription (#951)

This patch implements:

1. Return the result of rcl_publisher_get_topic_name() directly, instead
of splitting it by "/" and returns the last part.

2. Use rcl_subscription_get_topic_name() to get the topic of a
subscription.

Meanwhile, this patch updates the following tests:

- test/test-node-oo.js
- test/test-node.js
- test/test-remapping.js

Fix: #949, #950
commit 2f1e4ba9 (2024-01-08): Leverage rcl functions to get topic of publisher/subscription (#951)

This patch implements:

1. Return the result of rcl_publisher_get_topic_name() directly, instead
of splitting it by "/" and returns the last part.

2. Use rcl_subscription_get_topic_name() to get the topic of a
subscription.

Meanwhile, this patch updates the following tests:

- test/test-node-oo.js
- test/test-node.js
- test/test-remapping.js

Fix: #949, #950