RobotWebTools/rclnodejs #958: Error when installing rclnodejs Ubuntu 22.04 ROS2 Humble Node Version V18.18.0

danialhawari, 2024-03-15:
**Description**
Hello as per the above title, I would like to install this package. I have sourced the ROS2 to the project file. I have no idea on how to fix this issue. Thank you for your support.

- Library Version: Latest 0.22.3
- ROS Version: Humble
- Platform / OS: Ubuntu 22.04.04
- Node Version: 18.18.0
- NPM Version: 10.10.0
- Device: AMD x64

**Steps To Reproduce**

npm i rclnodejs

**Expected Behavior**
SUCCESS!

**Actual Behavior**
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm ERR! code 1
npm ERR! path /home/dani/Desktop/rcl-test/node_modules/rclnodejs
npm ERR! command failed
npm ERR! command sh -c npm run rebuild
npm ERR! > rclnodejs@0.26.1 rebuild
npm ERR! > npm run clean && node-gyp -j 16 rebuild
npm ERR! 
npm ERR! 
npm ERR! > rclnodejs@0.26.1 clean
npm ERR! > node-gyp clean && rimraf ./generated
npm ERR! 
npm ERR! 
npm ERR! make: Entering directory '/home/dani/Desktop/rcl-test/node_modules/rclnodejs/build'
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/addon.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/executor.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/handle_manager.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/rcl_action_bindings.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/rcl_bindings.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/rcl_handle.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/rcl_lifecycle_bindings.o
npm ERR!   CXX(target) Release/obj.target/rclnodejs/src/rcl_utilities.o
npm ERR! make: Leaving directory '/home/dani/Desktop/rcl-test/node_modules/rclnodejs/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@10.0.1
npm ERR! gyp info using node@18.18.0 | linux | x64
npm ERR! gyp info ok 
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@10.0.1
npm ERR! gyp info using node@18.18.0 | linux | x64
npm ERR! gyp info find Python using Python version 3.10.12 found at "/usr/bin/python3"
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args '/home/dani/.nvm/versions/node/v18.18.0/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args 'binding.gyp',
npm ERR! gyp info spawn args '-f',
npm ERR! gyp info spawn args 'make',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/home/dani/Desktop/rcl-test/node_modules/rclnodejs/build/config.gypi',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/home/dani/.nvm/versions/node/v18.18.0/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/home/dani/.cache/node-gyp/18.18.0/include/node/common.gypi',
npm ERR! gyp info spawn args '-Dlibrary=shared_library',
npm ERR! gyp info spawn args '-Dvisibility=default',
npm ERR! gyp info spawn args '-Dnode_root_dir=/home/dani/.cache/node-gyp/18.18.0',
npm ERR! gyp info spawn args '-Dnode_gyp_dir=/home/dani/.nvm/versions/node/v18.18.0/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args '-Dnode_lib_file=/home/dani/.cache/node-gyp/18.18.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args '-Dmodule_root_dir=/home/dani/Desktop/rcl-test/node_modules/rclnodejs',
npm ERR! gyp info spawn args '-Dnode_engine=v8',
npm ERR! gyp info spawn args '--depth=.',
npm ERR! gyp info spawn args '--no-parallel',
npm ERR! gyp info spawn args '--generator-output',
npm ERR! gyp info spawn args 'build',
npm ERR! gyp info spawn args '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build', '--jobs', 16 ]
npm ERR! g++: fatal error: cannot execute ‘/usr/lib/gcc/x86_64-linux-gnu/11/cc1plus’: execv: Argument list too long
npm ERR! compilation terminated.
npm ERR! g++: fatal error: cannot execute ‘/usr/lib/gcc/x86_64-linux-gnu/11/cc1plus’: execv: Argument list too long
npm ERR! compilation terminated.
npm ERR! make: *** [rclnodejs.target.mk:10632: Release/obj.target/rclnodejs/src/addon.o] Error 1
npm ERR! make: *** Waiting for unfinished jobs....
npm ERR! make: *** [rclnodejs.target.mk:10632: Release/obj.target/rclnodejs/src/executor.o] Error 1
npm ERR! g++: fatal error: cannot execute ‘/usr/lib/gcc/x86_64-linux-gnu/11/cc1plus’: execv: Argument list too long
npm ERR! compilation terminated.
npm ERR! make: *** [rclnodejs.target.mk:10632: Release/obj.target/rclnodejs/src/handle_manager.o] Error 1
npm ERR! g++: fatal error: cannot execute ‘/usr/lib/gcc/x86_64-linux-gnu/11/cc1plus’: execv: Argument list too long
npm ERR! compilation terminated.
npm ERR! make: *** [rclnodejs.target.mk:10632: Release/obj.target/rclnodejs/src/rcl_action_bindings.o] Error 1
npm ERR! g++: fatal error: cannot execute ‘/usr/lib/gcc/x86_64-linux-gnu/11/cc1plus’: execv: Argument list too long
npm ERR! compilation terminated.
npm ERR! make: *** [rclnodejs.target.mk:10632: Release/obj.target/rclnodejs/src/rcl_bindings.o] Error 1
npm ERR! g++: fatal error: cannot execute ‘/usr/lib/gcc/x86_64-linux-gnu/11/cc1plus’: execv: Argument list too long
npm ERR! compilation terminated.
npm ERR! make: *** [rclnodejs.target.mk:10632: Release/obj.target/rclnodejs/src/rcl_handle.o] Error 1
npm ERR! g++: fatal error: cannot execute ‘/usr/lib/gcc/x86_64-linux-gnu/11/cc1plus’: execv: Argument list too long
npm ERR! compilation terminated.
npm ERR! make: *** [rclnodejs.target.mk:10632: Release/obj.target/rclnodejs/src/rcl_lifecycle_bindings.o] Error 1
npm ERR! g++: fatal error: cannot execute ‘/usr/lib/gcc/x86_64-linux-gnu/11/cc1plus’: execv: Argument list too long
npm ERR! compilation terminated.
npm ERR! make: *** [rclnodejs.target.mk:10632: Release/obj.target/rclnodejs/src/rcl_utilities.o] Error 1
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack at ChildProcess.<anonymous> (/home/dani/.nvm/versions/node/v18.18.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:209:23)
npm ERR! gyp ERR! System Linux 6.5.0-21-generic
npm ERR! gyp ERR! command "/home/dani/.nvm/versions/node/v18.18.0/bin/node" "/home/dani/.nvm/versions/node/v18.18.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "-j" "16" "rebuild"
npm ERR! gyp ERR! cwd /home/dani/Desktop/rcl-test/node_modules/rclnodejs
npm ERR! gyp ERR! node -v v18.18.0
npm ERR! gyp ERR! node-gyp -v v10.0.1
npm ERR! gyp ERR! not ok

minggangw, 2024-03-18:
Hi @danialhawari Thanks for your feedback, what's your `g++` version? I tested with g++ 11.4.0 & ubuntu 22.04 and it works fine.

danialhawari, 2024-03-18:
> Hi @danialhawari Thanks for your feedback, what's your `g++` version? I tested with g++ 11.4.0 & ubuntu 22.04 and it works fine.

The G++ version is 11.4.0

minggangw, 2024-03-18:
Please run
```bash
npm install --verbose
```
to see the args, looks like
```bash
  g++ -o Release/obj.target/rclnodejs/src/executor.o ../src/executor.cpp '-DNODE_GYP_MODULE_NAME=rclnodejs' '-DUSING_UV_SHARED=1' '-DUSING_V8_SHARED=1' '-DV8_DEPRECATION_WARNINGS=1' '-D_GLIBCXX_USE_CXX11_ABI=1' '-D_LARGEFILE_SOURCE' '-D_FILE_OFFSET_BITS=64' '-D__STDC_FORMAT_MACROS' '-DOPENSSL_NO_PINSHARED' '-DOPENSSL_THREADS' '-DROS_VERSION=2205' '-DOS_LINUX' '-DBUILDING_NODE_EXTENSION' -I/home/minggang/.cache/node-gyp/20.11.0/include/node -I/home/minggang/.cache/node-gyp/20.11.0/src -I/home/minggang/.cache/node-gyp/20.11.0/deps/openssl/config -I/home/minggang/.cache/node-gyp/20.11.0/deps/openssl/openssl/include -I/home/minggang/.cache/node-gyp/20.11.0/deps/uv/include -I/home/minggang/.cache/node-gyp/20.11.0/deps/zlib -I/home/minggang/.cache/node-gyp/20.11.0/deps/v8/include -I../. -I../node_modules/nan -I/home/minggang/Downloads/humble/ros2-linux/include -I/home/minggang/Downloads/humble/ros2-linux/include/rcl -I/home/minggang/Downloads/humble/ros2-linux/include/rcutils -I/home/minggang/Downloads/humble/ros2-linux/include/rmw -I/home/minggang/Downloads/humble/ros2-linux/include/rcl_yaml_param_parser -I/home/minggang/Downloads/humble/ros2-linux/include/rosidl_typesupport_interface -I/home/minggang/Downloads/humble/ros2-linux/include/rcl_action -I/home/minggang/Downloads/humble/ros2-linux/include/action_msgs -I/home/minggang/Downloads/humble/ros2-linux/include/service_msgs -I/home/minggang/Downloads/humble/ros2-linux/include/unique_identifier_msgs -I/home/minggang/Downloads/humble/ros2-linux/include/builtin_interfaces -I/home/minggang/Downloads/humble/ros2-linux/include/rcl_lifecycle -I/home/minggang/Downloads/humble/ros2-linux/include/lifecycle_msgs -I/home/minggang/Downloads/humble/ros2-linux/include/rosidl_runtime_c -I/home/minggang/Downloads/humble/ros2-linux/include/rosidl_dynamic_typesupport -I/home/minggang/Downloads/humble/ros2-linux/include/type_description_interfaces  -fPIC -pthread -Wall -Wextra -Wno-unused-parameter -fstack-protector-strong -fPIE -fPIC -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Wextra -Wno-cast-function-type -m64 -O3 -fno-omit-frame-pointer -fno-rtti -std=gnu++17 -std=c++17 -MMD -MF ./Release/.deps/Release/obj.target/rclnodejs/src/executor.o.d.raw   -c
```

danialhawari, 2024-03-18:
`npm install --verbose`

> npm verb cli /home/dani/.nvm/versions/node/v18.18.0/bin/node /home/dani/.nvm/versions/node/v18.18.0/bin/npm
> npm info using npm@10.5.0
> npm info using node@v18.18.0
> npm verb title npm install
> npm verb argv "install" "--loglevel" "verbose"
> npm verb logfile logs-max:10 dir:/home/dani/.npm/_logs/2024-03-18T10_25_46_288Z-
> npm verb logfile /home/dani/.npm/_logs/2024-03-18T10_25_46_288Z-debug-0.log
> 
> up to date, audited 1 package in 158ms
> 
> found 0 vulnerabilities
> npm verb exit 0
> npm info ok

minggangw, 2024-03-19:
@danialhawari Please check your `AMENT_PREFIX_PATH` env var, someone ever met this issue due to super long paths.
See details https://github.com/RobotWebTools/rclnodejs/issues/880

danialhawari, 2024-03-22:
@minggangw Hello, thank you for your suggestion. I have fixed my issue by reinstalling ROS2 Humble.

closed 2024-03-22 by danialhawari

minggangw, 2024-03-25:
Good to know:)