RobotWebTools/rclnodejs #959: rclnodejs generate-ros-messages ROS2-Shared memory

Tombliboo96, 2024-04-17:
**Description**
Hi, I am using npx generate-ros-messages to parse the idl files in 
https://github.com/ZhenshengLee/ros2_shm_msgs 
and run a nodejs project. However, when running the project, there is an error in the generated js file under node_modules/rclnodejs/generated/ due to incorrect field names parsed by 
deallocator.freeStructMember(refObject, UInt8Wrapper.refObjectType, 'size'); 
I found that only the data field is returned in Uint8Wrapper while idl files also contain both data and size fields.

- Library Version: nodejs 18.19.1/20.11.1
- ROS Version: ROS2 humble /rclnodejs 0.26.0
- Platform / OS:

**Steps To Reproduce**

I use [https://github.com/ZhenshengLee/ros2_shm_msgs ](https://github.com/ZhenshengLee/ros2_shm_msgs) project to send ImageData to connect with rclnodejs, deploy the msg in the project to the system, and use the generate-ros-messages in rclnodejs for js conversion. Nodejs receives data through rosSubscriber

**Expected Behavior**
I want to connect to rclnodejs using shared memory in ROS2, not sure if it is feasible

**Actual Behavior**

minggangw, 2024-04-17:
Hi @Tombliboo96 thanks for your feedback, `rclnodejs` leverages `v8::ArrayBuffer` to eliminate additional memory copy when receiving messages from `rcl`, and it only works for the array of type:

```
['int8', 'uint8', 'int16', 'uint16', 'int32', 'uint32', 'float64', 'float32', 'char', 'byte']
```

Not sure it's what you called "shared memory", so do you meet any errors, like crashing?

Tombliboo96, 2024-04-17:
Hi, @minggangw  I got an error when using generate-ros-messages to convert idl file to js file and apply,
The content of the IDL file is as follows:
module shm_msgs {
  module msg {
    typedef uint8 uint8__256[256];
    module String_Constants {
      const uint8 MAX_SIZE = 255;
    };
    struct String {
      uint8__256 data;

      @default (value=0)
      uint8 size;
    };
  };
};
Viewing the generated js file found that the problem in the last uint8 size field cannot be passed through the system's built-in std_Uint8

minggangw, 2024-04-18:
Currently, rclnodejs doesn't support `.idl` files, are you sure the JS files are generated accordingly? Also, please check with [#764](https://github.com/RobotWebTools/rclnodejs/issues/764)

Tombliboo96, 2024-04-18:
I use the generate-ros-messages tool described in [https://github.com/RobotWebTools/rclnodejs ](https://github.com/RobotWebTools/rclnodejs)to generate the corresponding js file for the IDL file, and found the generated js file in /node_modules/rclnodejs/generated/, but it cannot be used normally。

Tombliboo96, 2024-04-18:
I use the generate-ros-messages tool described in [https://github.com/RobotWebTools/rclnodejs ](https://github.com/RobotWebTools/rclnodejs)to generate the corresponding js file for the IDL file, and found the generated js file in /node_modules/rclnodejs/generated/, but it cannot be used normally;
The content of the IDL file is as follows:
`module shm_msgs {
module msg {
typedef uint8 uint8__256[256];
module String_Constants {
const uint8 MAX_SIZE = 255;
};
struct String {
uint8__256 data;

  @default (value=0)
  uint8 size;
};
};
};`
指出的错误在识别上述字段中的size时出现；

minggangw, 2024-04-18:
Would you share your `.msg` file if you have?

Tombliboo96, 2024-04-18:
char[256] data
uint8 size 0
uint8 MAX_SIZE=255

minggangw, 2024-04-19:
So I will close this issue, thanks!

closed 2024-04-19 by minggangw