RobotWebTools/rclnodejs #970: Building Prod App explodes with invalid str int comparison in binding.gyp on Ubuntu

supertick, 2024-05-29:
**Description**
While attempting to do a production build with rclnodejs in an Electron App.
I get the following error.
` gyp info spawn args ]
    Traceback (most recent call last):
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py", line 45, in <module>
        sys.exit(gyp.script_main())
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 688, in script_main
        return main(sys.argv[1:])
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 680, in main
        return gyp_main(args)
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 647, in gyp_main
        [generator, flat_list, targets, data] = Load(
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 157, in Load
        result = gyp.input.Load(
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 2987, in Load
        LoadTargetBuildFile(
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 427, in LoadTargetBuildFile
        ProcessVariablesAndConditionsInDict(
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 1401, in ProcessVariablesAndConditionsInDict
        ProcessVariablesAndConditionsInList(value, phase, variables, build_file)
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 1414, in ProcessVariablesAndConditionsInList
        ProcessVariablesAndConditionsInDict(item, phase, variables, build_file)
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 1375, in ProcessVariablesAndConditionsInDict
        ProcessConditionsInDict(the_dict, phase, variables, build_file)
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 1246, in ProcessConditionsInDict
        ProcessVariablesAndConditionsInDict(
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 1375, in ProcessVariablesAndConditionsInDict
        ProcessConditionsInDict(the_dict, phase, variables, build_file)
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 1239, in ProcessConditionsInDict
        merge_dict = EvalCondition(
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 1155, in EvalCondition
        result = EvalSingleCondition(
      File "/home/gperry/.nvm/versions/node/v21.7.2/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 1184, in EvalSingleCondition
        if eval(ast_code, env, variables):
      File "<string>", line 1, in <module>
    TypeError: '>' not supported between instances of 'str' and 'int' while trying to load binding.gyp
    gyp ERR! configure error `

Looking at the binding.gyp file, exploding comparison between str and int.
Which looks like line 124.

`            'conditions': [
              [ 
                'ros_version > 2105', # Humble, Rolling, ... TODO - not tested due to broken setup_ros v3.3 action on windows 
                {
                  'include_dirs': 
`
**ros_version > 2105**

- Library Version: "rclnodejs": "^0.26.1",
- ROS Version: Humble ROS2
- Platform / OS: Ubuntu 22.04

**Steps To Reproduce**
Create an Electron App, add rclnodejs with yarn, build a Linux app

**Expected Behavior**
Not exploding

**Actual Behavior**
Compilation explodes

minggangw, 2024-05-30:
Thanks for your feedback, the `ros_version` is got from https://github.com/RobotWebTools/rclnodejs/blob/2d3ccaabeda7528c72cc766bd18f662530173351/lib/distro.js#L20-L29
So it's not a string actually, do you `source` your `setup.bash` already?

minggangw, 2024-06-06:
@supertick do you still have the issue?

minggangw, 2024-09-26:
@supertick please check with rclnodejs Electron demo just added https://github.com/RobotWebTools/rclnodejs/tree/develop/electron_demo

closed 2024-09-26 by minggangw