RobotWebTools/rclnodejs #973: Update README for Jazzy release

minggangw, 2024-05-31:


IoTDan, 2024-06-05:
Hello @minggangw ! i see a lot of activity for a Jazzy release. Thank you for this. I wanted to know if you have a general ETA for this? I realize it is best effort. Just looking to see scale of time - days, weeks, months. thank you!

minggangw, 2024-06-06:
Hi @IoTDan the latest release v0.27.0 has supported Jazzy https://github.com/RobotWebTools/rclnodejs/releases/tag/0.27.0, please have a try. I'm going to complete this task as it's done, thanks!

closed 2024-06-06 by minggangw

IoTDan, 2024-06-06:
Thank you!!! We tried this before last week, so perfect timing!! We will try this again with the v0.27 release.

Thanks,
Dan

You are receiving this email because you were mentioned.

minggangw, 2024-06-06:
Your feedback is welcome!