RobotWebTools/rclnodejs #975: Image Topic A web video server

kavikode, 2024-06-17:
Is it possible to stream a image topic on a webpage using rclnodejs? Any examples please to get the camera view from a robot displayed on the webpage?

minggangw, 2024-06-17:
You cannot run `rclnodejs` in a browser because nodejs is not supported, instead, you could:

1. Leverage [Electron](https://www.electronjs.org/) which support nodejs to develop desktop apps.
2. Leverage something like [web video server](https://github.com/RobotWebTools/web_video_server), but I'm not sure if it can work for ROS2.
3. Leverage [roslibjs](https://github.com/RobotWebTools/roslibjs) that can run in a browser to receive ROS2 topics.

chfritz, 2024-06-17:
Hi @kavikode, I strongly recommend learning about webrtc. I recently wrote a blog post on the various approaches people have used for streaming video to the web and why webrtc is the way to go: https://transitiverobotics.com/blog/streaming-video-from-robots.

kavikode, 2024-06-18:
thank you so much @minggangw and @chfritz! I will try the provided solutions. I appreciate your time to help me.