RobotWebTools/rclnodejs #979: Static array of bool serialisation error

Sata51, 2024-07-17:
**Description**

I've a message that contains a static array of bool. The advertised topic only received a static array of the right size with falsy value.

- Library Version: 0.26.1
- ROS Version: galactic
- Platform / OS: Jetson Orin (Ubuntu 20.04.6 LTS)

**Steps To Reproduce**

Create a message that contains something like this: 

```plaintext
bool[5] limited
bool[] unlimited
```

```typescript
const myArray = [true, false, true, false, false]
const msg = rclnodejs.createMessageObject('custom/msg/arrayMsg')
msg.limited = Array.from(myArray)
msg.unlimited = Array.from(myArray)

publisher.publish(msg);
```

**Expected Behavior**

```shell
❯ ros2 topic echo /arrayMsgTopic
limited:
- true
- false
- true
- false
- true
unlimited:
- true
- false
- true
- false
- true
```

**Actual Behavior**

```shell
❯ ros2 topic echo /arrayMsgTopic
limited:
- false
- false
- false
- false
- false
unlimited:
- true
- false
- true
- false
- true
```

minggangw, 2024-07-18:
@Sata51 thanks for your feedback, I will take a look.

closed 2024-08-06 by minggangw

minggangw, 2024-08-06:
@Sata51 The issue has been fixed on latest [0.27.2](https://github.com/RobotWebTools/rclnodejs/releases/tag/0.27.2), please try with it, thanks!

PR #984 (2024-08-06): Fix publishing a fixed array in a topic
Currently, when publishing a topic that contains a fixed array, the sub cannot receive the topic as expected values.

This patch fixes this issue for publishing a fixed array in a topic. Meanwhile, a unit test is added to ensure it works for different types of array.

Fix: #979
commit f9ea71b8 (2024-08-06): Fix publishing a fixed array in a topic (#984)

Currently, when publishing a topic that contains a fixed array, the sub
cannot receive the topic as expected values.

This patch fixes this issue for publishing a fixed array in a topic.
Meanwhile, a unit test is added to ensure it works for different types
of array.

Fix: #979
commit bfb8ed6b (2024-08-06): Fix publishing a fixed array in a topic (#984)

Currently, when publishing a topic that contains a fixed array, the sub
cannot receive the topic as expected values.

This patch fixes this issue for publishing a fixed array in a topic.
Meanwhile, a unit test is added to ensure it works for different types
of array.

Fix: #979