RobotWebTools/rclnodejs #992: rcl_timer_init2 not supported in Iron

dnikitop, 2024-08-12:
**Description**
On Iron, rclnodejs install fails due to depending on rcl_timer_init2, which isn't released for iron.
https://docs.ros.org/en/iron/p/rcl/search.html?q=rcl_timer_init2&check_keywords=yes&area=default#

This recent commit should be changed to be >2305 instead.
https://github.com/RobotWebTools/rclnodejs/commit/7cd08c2ef4fb598507c8cceac51f67610c0c4679
- Library Version: 0.27.3
- ROS Version: Iron
- Platform / OS: Ubuntu 22.04

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

minggangw, 2024-08-13:
Woops, I forgot the iron and I will prepare a hotfix version for this issue, thanks for your feedback!

closed 2024-08-13 by minggangw

minggangw, 2024-08-13:
Please try with https://github.com/RobotWebTools/rclnodejs/releases/tag/0.27.4

PR #993 (2024-08-13): Leverage rcl_timer_init2 since Jazzy
Fix: #992
commit 99fe923d (2024-08-13): Leverage rcl_timer_init2 since Jazzy (#993)

Fix: #992
commit f7fdd235 (2024-08-13): Leverage rcl_timer_init2 since Jazzy (#993)

Fix: #992