RobotWebTools/rclnodejs #1013: Typescript: `message` type on `SubscriptionCallback<T>` based on `isRaw`

mrjogo, 2024-12-26:
**Description**

`createSubscription()` takes a callback of type `SubscriptionCallback<T>`, which is `(message: MessageType<T> | Buffer) => void;`. However, my understanding is `message` for a given callback will actually only be one of `MessageType<T>` or `Buffer`, depending on the value of the option `isRaw` passed to `createSubscription()`.

Therefore, instead of having `message` be the union of `MessageType<T> | Buffer`, there should be two `SubscriptionCallback` types (or a generic), one with `message: MessageType<T>` and one with `message: Buffer`, and the correct one is used depending on the `isRaw` option value.

The workaround for the current implementation is to add discriminator checks at the top of all callbacks:

```typescript
if (Buffer.isBuffer(message)) {
  throw new Error("This should never be a Buffer")
}
```

- Library Version: 0.27.5
- ROS Version: Jazzy
- Platform / OS: Ubuntu 22.04

**Steps To Reproduce**

```typescript
node.createSubscription(
      "std_msgs/msg/String",
      "/my_topic",
      {
        isRaw: false,
      },
      (message) => {
        // Gives error:
        // Type 'Buffer<ArrayBufferLike> | String' is not assignable to type 'String'.
        // Property 'data' is missing in type 'Buffer<ArrayBufferLike>' but required in type 'String'.ts(2322)
        const myString: rclnodejs.std_msgs.msg.String = message;
      }
    );
```

**Expected Behavior**

No Typescript error.

**Actual Behavior**

Typescript error, or discriminator required.

closed 2025-02-08 by minggangw

minggangw, 2025-02-08:
Hi @mrjogo please try with latest 0.29.0, thanks!

mrjogo, 2025-02-10:
Works great, thanks!

PR #1028 (2025-02-08): [TypeScript] Provide two types of callback for a subscription
Currently, the type of callback for a subscription is defined as:
```ts
type SubscriptionCallback<T extends TypeClass<MessageTypeClassName>> =
  (message: MessageType<T> | Buffer) => void
```
which causes a type error when pass a callback either for a normal message or raw message.

This patch provides two types of callback for a subscription, which are:

1. For a normal message
```ts
type SubscriptionCallback<T extends TypeClass<MessageTypeClassName>> =
  (message: MessageType<T>) => void;
``` 

2. For a raw message:
```ts
type Sub [... cut]
commit 3ffc5439 (2025-02-08): [TypeScript] Provide two types of callback for a subscription (#1028)

Currently, the type of callback for a subscription is defined as:
```ts
type SubscriptionCallback<T extends TypeClass<MessageTypeClassName>> =
  (message: MessageType<T> | Buffer) => void
```
which causes a type error when pass a callback either for a normal message or raw message.

This patch provides two types of callback for a subscription, which are:

1. For a normal message
```ts
type SubscriptionCallback<T extends TypeClass<MessageTypeClassName>> =
  (message: MessageType<T>) => void;
``` 

2. For a raw message:
```ts [... cut]
commit 72b40b01 (2025-02-08): [TypeScript] Provide two types of callback for a subscription (#1028)

Currently, the type of callback for a subscription is defined as:
```ts
type SubscriptionCallback<T extends TypeClass<MessageTypeClassName>> =
  (message: MessageType<T> | Buffer) => void
```
which causes a type error when pass a callback either for a normal message or raw message.

This patch provides two types of callback for a subscription, which are:

1. For a normal message
```ts
type SubscriptionCallback<T extends TypeClass<MessageTypeClassName>> =
  (message: MessageType<T>) => void;
``` 

2. For a raw message:
```ts [... cut]