RobotWebTools/rclnodejs #1024: Client waitForService response can be incorrect

PierreSachot, 2025-01-29:
**Description**
ROS node can implement a service but throw an exception while calling it. The issue is that this error is never thrown on rclnodejs side (C++ I guess) and the `client.sendRequest` never response.

- Library Version: 0.27.4
- ROS Version: Ros2 humble
- Platform / OS: Ubuntu 22

**Steps To Reproduce**

Inside our code base, we have a node named `/bt_navigator_rclcpp_node`. The issue is that we do not spin the node but it is executed by another node.
When calling `ros2 param list /bt_navigator_rclcpp_node` inside a terminal, the following exception is thrown:
```bash
Exception while calling service of node '/bt_navigator_rclcpp_node': None
``` 

**Actual Behavior**
In my Node code, I use the following function in order to call the service:
```TypeScript
private _callService(client: Client<ROS_CLIENT_TYPES>, request: ROS_REQUEST_TYPES, callback: (response?: unknown) => void): void {
    const doesServiceExists = client.isServiceServerAvailable();
    if (!doesServiceExists) {
      callback(undefined);
      this._destroyClient(client);
      return;
    }

    client.waitForService(ROS_SUBSCRIBER_TIMEOUT_MS)
      .then((isAvailable: boolean) => {
        if (!isAvailable) {
          callback(undefined);
          this._destroyClient(client);
        }
        console.log('1. Calling: service', client.serviceName);
        client.sendRequest(request, (response: unknown) => {
          console.log('2. Response received: service', client.serviceName);
          callback(response);
          this._destroyClient(client);
        });
      });
  }
```

The issue is that the rclnode `isServiceServerAvailable` function returns true (which means that the list parameter service exists on this node) and `waitForService` also returns true (which means that the service is available right now). Next, the first console log is properly printed, but the second one is never displayed. I also tried to catch any exception by surrounding the `sendRequest` method with a try catch. No error are thrown on rclnodejs side.

**Expected Behavior**
`waitForService` should returns false in case of unavailable service.

minggangw, 2025-02-05:
Hi @PierreSachot thanks for your detailed description and investigation, so your scenario is:

1. A node named `bt_navigator_rclcpp_node`, running the actual service.
2. The other node implemented by `rclnodejs` running the code snippet shown in your comment.
3. Running `ros2 param list /bt_navigator_rclcpp_node`, then get the exception.

If I understand correctly, `isServiceServerAvailable()` returns the service (under node `bt_navigator_rclcpp_node`) status of step 1, instead of the list parameter service, please also check with https://robotwebtools.github.io/rclnodejs/docs/0.28.0/Client.html

PierreSachot, 2025-04-14:
Sorry @minggangw I missed your answer. 
Exactly, this is the proper steps to reproduce. The issue is related to `waitForService` and not `isServiceServerAvailable`. Did you manage to reproduce and/or correct it?

minggangw, 2025-04-15:
I see, I simply tried with [client-example.js](https://github.com/RobotWebTools/rclnodejs/blob/develop/example/client-example.js) and the `waitForService` seems working as expected, can you please try it again using latest version of rclnodejs?

PierreSachot, 2025-06-03:
@minggangw Hi, thanks for your response.

I did try again using `rlnodejs 0.27.4`. I can confirm that the issue is still present. 
We are facing it once again in another context:
1. Create a first node
2. Create an rclnodejs node
3. Create a client to talk with the first node inside the rclnodejs node
4. Send a first request: the response is received and everything works well
5. Kill the first node
6. Send a second request: `waitForService` response is resolving and when sending a second request to the node, there is no response / no error catched.

Thanks a lot in advance,
Pierre

minggangw, 2025-06-10:
Hi @PierreSachot I did some simple testing as

1. ROS2 Jazzy + Ubuntu 24.04
2. rclnodejs v1.2.0
3. [client-example.js](https://github.com/RobotWebTools/rclnodejs/blob/develop/example/client-example.js) + [service-example.js](https://github.com/RobotWebTools/rclnodejs/blob/develop/example/service-example.js)

The screen recording:

[image]

Please help to check if the testing steps is what you described above, thanks!

PierreSachot, 2026-09-29:
Dear @minggangw,

Thanks a lot for the time you took. I did upgrade to rclnodejs v2.2.0 and congratulations, the issue is fixed. Everything works as expected!

I close the issue. Regards,
Pierre

closed 2026-09-29 by PierreSachot

minggangw, 2026-09-29:
> Dear [@minggangw](https://github.com/minggangw),
> 
> Thanks a lot for the time you took. I did upgrade to rclnodejs v2.2.0 and congratulations, the issue is fixed. Everything works as expected!
> 
> I close the issue. Regards, Pierre

Thanks for your confirmation, good to know the issue is gone!

PR #1027 (2025-02-07): Add types to ParameterService
Since ROS2 Jazzy, parameter services support a service type of `rcl_interfaces/srv/GetParameterTypes`, which can get a list of parameter types associated with the requested parameters.

This patch implements:

1. Add service `/node_name/get_parameters`, which is used to get the parameter types.
2. Add unit test, `Get_parameter_types`, into test-parameter-service.js.

Meanwhile, this patch fixes the timeout issue when executing:
```bash
ros2 param list node_name
```
Due to absence of `/node_name/get_parameters` service.

**Reference**

[1] https://docs.ros.org/en/j [... cut]
commit e99fb8e1 (2025-02-07): Add types to ParameterService (#1027)

Since ROS2 Jazzy, parameter services support a service type of `rcl_interfaces/srv/GetParameterTypes`, which can get a list of parameter types associated with the requested parameters.

This patch implements:

1. Add service `/node_name/get_parameters`, which is used to get the parameter types.
2. Add unit test, `Get_parameter_types`, into test-parameter-service.js.

Meanwhile, this patch fixes the timeout issue when executing:
```bash
ros2 param list node_name
```
Due to absence of `/node_name/get_parameters` service.

**Reference**

[1] https://docs.ros [... cut]
commit 25811023 (2025-02-08): Add types to ParameterService (#1027)

Since ROS2 Jazzy, parameter services support a service type of `rcl_interfaces/srv/GetParameterTypes`, which can get a list of parameter types associated with the requested parameters.

This patch implements:

1. Add service `/node_name/get_parameters`, which is used to get the parameter types.
2. Add unit test, `Get_parameter_types`, into test-parameter-service.js.

Meanwhile, this patch fixes the timeout issue when executing:
```bash
ros2 param list node_name
```
Due to absence of `/node_name/get_parameters` service.

**Reference**

[1] https://docs.ros [... cut]