RobotWebTools/rclnodejs #1025: Subscription callback integer type is incorrect if integer_value > 9999999999999999

PierreSachot, 2025-01-29:
**Description**
When subscribing to a parameter event in Node.js, if the data type for the parameter is set to `Integer` but the assigned value exceeds `9999999999999999` (15 digits), its type changes unexpectedly from `'number'` to `'string'`. This behavior causes the `Parameter.fromParameterMessage` function to fail because it expects an integer, not a string.

This issue seems to be related to the following previously reported issues:  
- https://github.com/RobotWebTools/rclnodejs/issues/103  
- https://github.com/RobotWebTools/rclnodejs/issues/836  

**Environment Details:**
- **Library Version:** `0.28.1`  
- **ROS Version:** ROS 2 Humble  
- **Platform/OS:** Ubuntu 22.04  

---

**Steps To Reproduce** 
1. Create a Python file with the following code to declare a parameter:
```python
 import rclpy
  import rclpy.node
  
  class MinimalParam(rclpy.node.Node):
      def __init__(self):
          super().__init__('minimal_param_node')
  
          self.declare_parameter('my_parameter', 1000)
  
  
  def main():
      rclpy.init()
      node = MinimalParam()
      rclpy.spin(node)
  
  if __name__ == '__main__':
      main()
```

2. Create the following Node.js code that subscribes to parameter events:
```javascript
// Create ROS Node
const node = new Node('TestNode', '', Context.defaultContext(), options);
node.spin();

node.createSubscription(
  'rcl_interfaces/msg/ParameterEvent',
  '/parameter_events',
  { qos: QoS.profileParameterEvents },
  (buff: any) => {
    const msg = buff as ParameterEvent;

    console.log('Received parameter event from node: ', msg.node);
    if (msg.changed_parameters.length !== 0) {
      console.log('msg.value = ', msg.changed_parameters[0].value.integer_value, typeof msg.changed_parameters[0].value.integer_value);
    }
  }
```

3. Start the Python node and the Node.js code in separate terminals
4. In a third terminal, update the parameter value to `9999999999999999` using the following command:
```bash
ros2 param set /minimal_param_node my_parameter 9999999999999999
```
You should in Node.js terminal that the parameter integer_value type is `number`, next type:

5. Now, update the parameter value to `10000000000000000` with the following command:
```bash
ros2 param set /minimal_param_node my_parameter 10000000000000000
```
Now, the parameter integer_value type is `string`

---
**Actual Behavior**
When the parameter value exceeds `9999999999999999`, its data type transitions from `number` to `string` in Node.js. This type mismatch causes the `Parameter.fromParameterMessage` function to fail, as it expects the value to remain an integer.

A possible workaround to handle this issue could look like the following code snippet:  
```javascript
try {
  convertedMsg = Parameter.fromParameterMessage(msg);
  return convertedMsg;
} catch (error) {
  // get type
  const parameterType = msg.value.type;
  // handle array errors
  if (parameterType === ParameterType.INTEGER) {
    msg.value.integer_value = Number(msg.value.integer_value);
    convertedMsg = Parameter.fromParameterMessage(msg);
  }
}
```
This workaround allows the application to recover from the type mismatch by explicitly casting the integer_value back to a number before processing the parameter message. However, it only addresses the symptom of the issue and does not resolve the root cause.

**Expected Behavior**
The casting behavior (converting a `string` back to `number` when `ParameterType.INTEGER` is detected) should be directly detected and implemented in the `fromParameterMessage` function. This would eliminate the need for external type-handling workarounds and ensure consistency regardless of the parameter value.

minggangw, 2025-02-05:
Hi @PierreSachot thanks for your investigation, as you said, it's due to the limitation of JavaScript when representing [integer](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Number/MAX_SAFE_INTEGER), [BigInt](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/BigInt/asIntN) may work for this case (suggested by #836), I haven't check with it. BTW, you are welcome to submit a PR to fix it.

minggangw, 2025-02-14:
Fixed by https://github.com/RobotWebTools/rclnodejs/pull/1030

closed 2025-02-14 by minggangw