RobotWebTools/rclnodejs #1058: Typo in TypeScript: registerOnDectivate

kissadam21, 2025-03-05:
Hi everyone,

When trying to make a TS lifecycle node, I noticed an issue. In typescript registerOnDeactivate is not available, we can only use registerOnDectivate. If I try to start the compiled package the following error appears:
```
[node-1] Error initializing LifecycleNode: TypeError: this.registerOnDectivate is not a function
```
After fixing the typo in types/lifecycle.d.ts line 250, everything works.

I want to ask for a fix for this, or I can also make a pull request if that's easier for you.
Thanks in advance!

minggangw, 2025-03-05:
Hi @kissadam21, thanks for your feedback, please go ahead to submit your PR :) thanks!

kissadam21, 2025-03-05:
Hey, I think the test lines are correct there; only the registerOnDeactivate was incorrect, which I couldn't find a testing part for. 

Here is my PR:
https://github.com/RobotWebTools/rclnodejs/pull/1059

minggangw, 2025-03-05:
PR merged, closed.

closed 2025-03-05 by minggangw