RobotWebTools/rclnodejs #1065: Rimraf error when installing rclnodejs

kissadam21, 2025-03-07:
Hey all,

With the new release, unfortunately the `npm install rclnodejs` fails; it needs the rimraf package for installation, which was moved to devDependencies in the latest release. The rimraf is still used in the clean script in package.json:22, so it makes the installation fail.

`70.71 npm ERR! > rclnodejs@0.32.0 rebuild
70.71 npm ERR! > npm run clean && node-gyp -j 16 rebuild
70.71 npm ERR! 
70.71 npm ERR! 
70.71 npm ERR! > rclnodejs@0.32.0 clean
70.71 npm ERR! > node-gyp clean && rimraf ./generated
70.71 npm ERR! gyp info it worked if it ends with ok
70.71 npm ERR! gyp info using node-gyp@9.1.0
70.71 npm ERR! gyp info using node@16.20.2 | linux | x64
70.71 npm ERR! gyp info ok 
70.71 npm ERR! sh: 1: rimraf: not found`

minggangw, 2025-03-08:
Hi @kissadam21 thanks for creating this issue, #1067 merged to fix it and I will prepare a hot-fix release together with #1066

closed 2025-03-08 by minggangw

kissadam21, 2025-03-10:
Hi @minggangw, thanks for the quick support!

PR #1067 (2025-03-08): Move rimraf to dependencies
This patch moves `rimraf` to `dependencies` in package.json as it's still needed. Meanwhile, add `npm run clean` as the last step for Github Actions to clean up the workspace for Linux platform.

Fix: #1065
commit 8152c140 (2025-03-08): Move rimraf to dependencies (#1067)

This patch moves `rimraf` to `dependencies` in package.json as it's still needed. Meanwhile, add `npm run clean` as the last step for Github Actions to clean up the workspace for Linux platform.

Fix: #1065
commit 90de7299 (2025-03-08): Move rimraf to dependencies (#1067)

This patch moves `rimraf` to `dependencies` in package.json as it's still needed. Meanwhile, add `npm run clean` as the last step for Github Actions to clean up the workspace for Linux platform.

Fix: #1065