RobotWebTools/rclnodejs #1089: Couldn't import the type definitions of std_msgs in my Typescript program

A-Marge-SSCI, 2025-03-25:
**Description**

I seem to be unable to import the type definitions of std_msgs in my Typescript program. I am new to typescript/javascript, but I think I've checked the obvious stuff:

My `tsconfig.json` file contains the following:
```
...
  "compilerOptions": {
   ...
    "module": "commonjs",
    "moduleResolution": "node",
    "target": "es6"
    },
...
```

I ran `npx generate-ros-messages` and in `<yourproject>/node_modules/rclnodejs/generated/` I see a `std_msgs` directory with files for every type, including strings.
I also have an `interfaces.d.ts` file located in `<yourproject>/node_modules/types`, and it contains:
```
...
declare module 'rclnodejs' {
...
  namespace std_msgs {
  ...
    namespace msgs {
    ...
      export interface String {
      ...
```

I'm just not sure why I can't access the `rclnodejs.std_msgs.msg.String` type directly, do I have to separately include interface.d.ts?

- Library Version:   "rclnodejs": "0.27.4"
- ROS Version: Ros2 - Humble
- Platform / OS: Description:    Ubuntu 20.04.6 LTS

**Steps To Reproduce**
1. Start the following typescript code using `npx tsx src/example.ts`
2. Send a ros2 message to the /msg_sub topic using `ros2 topic pub /msg_sub std_msgs/msg/String "data: hello"`

```typescript
import * as rclnodejs from 'rclnodejs';

async function main() {
  await rclnodejs.init();

  const node = new rclnodejs.Node('ros2_pub_ex');
  const publisher = node.createPublisher('std_msgs/msg/String', 'msg_pub');
  node.createSubscription('std_msgs/msg/String', 'msg_sub', (msg) => {
    if ( msg instanceof rclnodejs.std_msgs.msg.String) {
      console.log(`Received std_msgs/msg/String with data: ${msg.data}`)
    }
  });
  node.spin();

  let num = 0;
  setInterval(() => {
      let stringMsgObject = rclnodejs.createMessageObject('std_msgs/msg/String');
      stringMsgObject.data = `hello world ${num++}`;
      console.log('publishing ', stringMsgObject);
      publisher.publish(stringMsgObject);
  }, 1000);
}

void main();

process.on('SIGINT', () => process.exit(1));
``` 

**Expected Behavior**

I expected a console printout of receiving the message. Ex:
`Received std_msgs/msg/String with data: hello`

**Actual Behavior**

```shell
/workspaces/rclnodejs_example/src/example.ts:10
    if ( msg instanceof rclnodejs.std_msgs.msg.String) {
                                           ^

TypeError: Cannot read properties of undefined (reading 'msg')
```

minggangw, 2025-03-26:
Please try below to see if it works for you
```js
if ( msg instanceof rclnodejs.require('std_msgs').msg.String) {
```

A-Marge-SSCI, 2025-03-26:
Thanks for the response! I updated that section of the code:

```typescript
  node.createSubscription('std_msgs/msg/String', 'msg_sub', (msg) => {
    console.log('msg recieved ',msg);
    if ( msg instanceof rclnodejs.require('std_msgs').msg.String) {
      console.log(`Received std_msgs/msg/String with data: ${msg.data}`)
    }
  });
```

And when I run the typescript compiler `npx tsc --project ./tsconfig.json` I get the following build error:

```shell

error TS2339: Property 'msg' does not exist on type 'object'.

10     if ( msg instanceof rclnodejs.require('std_msgs').msg.String) {
                                                         ~~~

error TS2339: Property 'data' does not exist on type 'Buffer<ArrayBufferLike> | String'.
  Property 'data' does not exist on type 'Buffer<ArrayBufferLike>'.

11       console.log(`Received std_msgs/msg/String with data: ${msg.data}`)
                                                                    ~~~~
```

The code still compiles to js, and I can run it with `node ./dist/example.js`, but if I run it and then in a separate terminal, run `ros2 topic pub /msg_sub std_msgs/msg/String "data: hello"`, I get the printout that the message is received
```
msg recieved  { data: 'hello' }
```
but not the printout about the data field
```
Received std_msgs/msg/String with data: ${msg.data}
```

I could work around the error by casting the type of `msg` as `any`, but that defeats the purpose of using typescript

A-Marge-SSCI, 2025-03-26:
I did some more work on this, and I can't tell the callback to expect the type `rclnodejs.std_msgs.msg.String` directly, resulting in the same error as before
```typescript
  node.createSubscription('std_msgs/msg/String', 'msg_sub', (msg:rclnodejs.std_msgs.msg.String) => {
    console.log('msg received ',msg);
    console.log(`Received std_msgs/msg/String with data: ${msg.data}`)
  });
```
```shell
error TS2345: Argument of type '(msg: rclnodejs.std_msgs.msg.String) => void' is not assignable to parameter of type 'SubscriptionCallback<"std_msgs/msg/String">'.
  Types of parameters 'msg' and 'message' are incompatible.
    Type 'Buffer<ArrayBufferLike> | String' is not assignable to type 'String'.
      Property 'data' is missing in type 'Buffer<ArrayBufferLike>' but required in type 'String'.
```

But, I was able to figure out that I could get the build error to go away if I first expect `msg` to be of type `rclnodejs.Message`, and then assert it to be of type `rclnodejs.std_msgs.msg.String`
```typescript
  node.createSubscription('std_msgs/msg/String', 'msg_sub', (msg:rclnodejs.Message) => {
    console.log('msg received ',msg);
    console.log(`Received std_msgs/msg/String with data: ${(msg as rclnodejs.std_msgs.msg.String).data}`)
  });
```

A-Marge-SSCI, 2025-03-26:
I did some more digging, and it seems like casting `msg` as type `rclnodejs.std_msgs.msg.String` within the callback is going to be the best solution for now.  Since `createSubscription` expects the callback to accept a union of types, my callback has to accept that union of types, i.e. `(msg:rclnodejs.Message)`. Perhaps a better long-term solution would be to have `createSubscription` allow for generics? That way, I could do something like this:

```typescript
  node.createSubscription<rclnodejs.std_msgs.msg.String>('std_msgs/msg/String', 'msg_sub', (msg) => {
    console.log('msg received ',str_msg);
    console.log(`Received std_msgs/msg/String with data: ${str_msg.data}`)
  });
```

Either way, it seems like the message type would have to be specified in two places: first as an argument into `createSubscription` (`std_msgs/msg/String`), and second to specify the typescript type of `msg` (`rclnodejs.std_msgs.msg.String`), and its up to users to make sure the two match.

minggangw, 2025-03-27:
Hi @A-Marge-SSCI thanks for your deep investigation, I notice the test below should cover your case
https://github.com/RobotWebTools/rclnodejs/blob/ff07c860caba762898bcfda3624a90edea03a32d/test/types/index.test-d.ts#L158-L165

As you said the callback is a union of two types, so I think either should work per your case.
https://github.com/RobotWebTools/rclnodejs/blob/ff07c860caba762898bcfda3624a90edea03a32d/types/node.d.ts#L322-L327

minggangw, 2025-03-27:
Please check with this similar issue https://github.com/RobotWebTools/rclnodejs/issues/1013

A-Marge-SSCI, 2025-03-27:
Ah, my mistake, I see what happened here. I'm using version 0.27.4, and issue https://github.com/RobotWebTools/rclnodejs/issues/1013 led to a fix in 0.29.0. However, when I try upgrading to version 0.29.0 (or any version >= 0.28.1), I get the following error when I try to run `npm install rclnodejs@0.29.0`

```shell
npm error g++: error: unrecognized command line option ‘-std=c++20’; did you mean ‘-std=c++2a’?
npm error make: *** [rclnodejs.target.mk:163: Release/obj.target/rclnodejs/src/addon.o] Error 1
```

This indicates that rclnodejs v0.28.1 or greater requires gcc10. I'm running this in a container running Ubuntu 20.04, which defaults to gcc version 9.4.0. So, either I try to install gcc10 into the container, or I upgrade the container to Ubuntu 22.04. I'll update once I've done that.

minggangw, 2025-03-28:
@A-Marge-SSCI The reason why we have to use C++20 standard is that it's needed for compiling with latest nodejs 23.x, please see details https://github.com/RobotWebTools/rclnodejs/pull/1020.

For your case, I suggest you could upgrade your container, please also check with the C++20 support against compilers https://en.cppreference.com/w/cpp/compiler_support/20, thanks!

A-Marge-SSCI, 2025-03-28:
Thank you - I understand. I'm working on upgrading my container now. Out of curiosity, is there a way to specify using a different compiler when installing rclnodejs via npm?

A-Marge-SSCI, 2025-03-28:
I was able to upgrade to v0.32.4 after updating my container to ubuntu 22.04, and verified that my problem went away. Looks like the takeaway here is that I had to make sure I was using the latest version of rclnodejs, which required me to update to ubuntu 22.04. 

I was previously unable to install the latest version of rclnodejs as it would crash mid-install. rclnodejs uses the `-std=c++20` command line parameter, but my container was running Ubuntu 20.04 and had gcc 9.4, which instead uses `-std=c++2a` to specify c++20 support.
For me, it was easiest to just update my container to use a newer version of ubuntu as opposed to downloading gcc 10 into the container and trying to figure out how to specify compiler versions when installing rclnodejs using npm.

Thanks for the help!

closed 2025-03-28 by A-Marge-SSCI

minggangw, 2025-03-31:
@A-Marge-SSCI let's make it simple, I submit a PR which enables `C++20` after nodejs 22, please try with https://github.com/RobotWebTools/rclnodejs/releases/tag/0.32.5 and enjoy your original container :)

PR #1092 (2025-03-31): Only enable C++20 when nodejs >= 23
This patch implements to enable C++20 only when nodejs >= 23, which improve the compatibility with g++ lower versions, because C++20 only needed for nodejs >= 23.

Fix: #1089
commit 4e6f4123 (2025-03-31): Only enable C++20 when nodejs >= 23 (#1092)

This patch implements to enable C++20 only when nodejs >= 23, which improve the compatibility with g++ lower versions, because C++20 only needed for nodejs >= 23.

Fix: #1089
commit c7dd6605 (2025-03-31): Only enable C++20 when nodejs >= 23 (#1092)

This patch implements to enable C++20 only when nodejs >= 23, which improve the compatibility with g++ lower versions, because C++20 only needed for nodejs >= 23.

Fix: #1089