RobotWebTools/rclnodejs #1094: Issue Generating ROS Messages

leslier7, 2025-04-01:
**Description**
I am running `npx generate-ros-messages` to generate my messages, but it is failing with a TypeError: Cannot read properties of null (reading '1') error.

- Library Version: v0.32.5
- ROS Version: Humble
- Platform / OS: Jetson Nano Ubuntu 22.04.5 LTS

**Steps To Reproduce**

I tried it on a different machine, which doesn't have the same problem, so I can't figure out what is different here. But running `npx generate-ros-messages` or `node <js_file>` causes the problem.

**Expected Behavior**
Generates the ROS messages

**Actual Behavior**
TypeError: Cannot read properties of null (reading '1')
    at getSubFolder (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:55:54)
    at grabInterfaceInfo (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:65:19)
    at addInterfaceInfos (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:161:25)
    at /home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:198:32
    at Array.map (<anonymous>)
    at findAmentPackagesInDirectory (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:198:14)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5)
    at async generateInPath (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/index.js:30:16)
    at async Object.generateAll (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/index.js:60:7)
    at async Object.init (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/index.js:275:5)

Every time I run `npx generate-ros-messages` it deletes everything in the `rclnodejs/generated` folder in the `node_modules`

minggangw, 2025-04-02:
Hi @leslier7 thanks for submitting the issue, I cannot repro it on my side. Would you please log the `filePath` (line 55) locally to see what's the value? It seems that you may install some customize messages which could cause kind of exception.
https://github.com/RobotWebTools/rclnodejs/blob/baba6695777efbcfbed9c7a50fab862f2baa0d91/rosidl_gen/packages.js#L49-L60

leslier7, 2025-04-02:
I added a few more console.log() statements, this time in https://github.com/RobotWebTools/rclnodejs/blob/develop/rosidl_gen/index.js#L59-L61.

I added
`try {
  console.log("Path: " + path);
  await generateInPath(path);
} catch (error) {
  console.error("Caught an error while generating in path:", error);
  console.error("Stack trace:", error.stack);
  // Check for null values and print them
    if (error.message.includes("Cannot read properties of null")) {
        console.error("Null value encountered. Path value:", path);
    }
}
`
And it returned
Caught an error while generating in path: TypeError: Cannot read properties of null (reading '1')
    at getSubFolder (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:60:54)
    at grabInterfaceInfo (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:70:19)
    at addInterfaceInfos (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:167:25)
    at /home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:204:32
    at Array.map (<anonymous>)
    at findAmentPackagesInDirectory (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:204:14)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5)
    at async generateInPath (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/index.js:30:16)
    at async Object.generateAll (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/index.js:62:3)
    at async main (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/scripts/generate_messages.js:27:5)
Stack trace: TypeError: Cannot read properties of null (reading '1')
    at getSubFolder (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:60:54)
    at grabInterfaceInfo (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:70:19)
    at addInterfaceInfos (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:167:25)
    at /home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:204:32
    at Array.map (<anonymous>)
    at findAmentPackagesInDirectory (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/packages.js:204:14)
    at process.processTicksAndRejections (node:internal/process/task_queues:105:5)
    at async generateInPath (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/index.js:30:16)
    at async Object.generateAll (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/rosidl_gen/index.js:62:3)
    at async main (/home/bad/Frontend/electron-communication/node_modules/rclnodejs/scripts/generate_messages.js:27:5)
Null value encountered. Path value: /opt/ros/humble

It seems like for some reason it is trying to look in the top level ROS install. I'll check my AMENT_PREFIX_PATH to see if something is weird there

leslier7, 2025-04-02:
I just re-installed ROS2 on my system, and now it works, so it must have been something on my end.

closed 2025-04-02 by leslier7