RobotWebTools/rclnodejs #1096: Can't figure out how to get list of service client or servers names and types by node

usmahm, 2025-04-06:
Is there a way to get list of service client or servers names and types by node, similar to how we can get list of publishers or subscribers names and node using these respective methods `getPublisherNamesAndTypesByNode, getSubscriptionNamesAndTypesByNode` ?

minggangw, 2025-04-07:
@usmahm thanks for your feedback, I think you are finding:
https://github.com/RobotWebTools/rclnodejs/blob/baba6695777efbcfbed9c7a50fab862f2baa0d91/lib/node.js#L977
Also check out the [document](https://robotwebtools.github.io/rclnodejs/docs/0.32.0/Node.html).

For the client equivalent, after checking with the latest [rcl](https://github.com/ros2/rcl) which is the middleware library `rclnodjs` builds on, there dose kind of method to be exposed, see below:

https://github.com/ros2/rcl/blob/cc72fc1769ce9e6557c0f01efde982b3212e77b0/rcl/include/rcl/graph.h#L254

I'm going to add the missing method for getting client, thanks!

usmahm, 2025-04-07:
@minggangw, yes thank you.

I saw how to get the service names and types using `getServiceNamesAndTypesByNode `, however, I couldn't find the pair that gets the client names and types, and I need both for what I am trying to implement.

When do you think we can get the method for getting client?
Also, are there any other ways I can achieve that pending the time that is available?

minggangw, 2025-04-07:
#1097 merged, a new release of `rclnodejs` will be followed later. You can try with below in advance.

```bash
$ npm install https://github.com/RobotWebTools/rclnodejs#develop
```

PR #1097 (2025-04-07): Support rcl_get_client_names_and_types_by_node
This patch implements to expose `rcl_get_client_names_and_types_by_node` provided by `rcl` library, including:

1. Expose `rcl_get_client_names_and_types_by_node` by binding to JavaScript method `GetClientNamesAndTypesByNode`.
2. Add the type definition for `GetClientNamesAndTypesByNode` into `rclnodejs.d.ts`.
3. Add unit test for type into `test/types/index.test-d.ts`.
4. Add example into `example/ros-graph-example.js`.

Fix: #1096
commit bc6d4e2f (2025-04-07): Support rcl_get_client_names_and_types_by_node (#1097)

This patch implements to expose `rcl_get_client_names_and_types_by_node` provided by `rcl` library, including:

1. Expose `rcl_get_client_names_and_types_by_node` by binding to JavaScript method `GetClientNamesAndTypesByNode`.
2. Add the type definition for `GetClientNamesAndTypesByNode` into `rclnodejs.d.ts`.
3. Add unit test for type into `test/types/index.test-d.ts`.
4. Add example into `example/ros-graph-example.js`.

Fix: #1096
closed 2025-04-07 by minggangw